OpenFTC/EasyOpenCV #1: Initialization stalls out

Matthew1231A, 2019-10-07:
Hey, 

In v1.1 of the project, our initialization of a webcam always "blue screens" and fails to load the webcam's image. Please let me know what supporting documentation I need to submit to help solve this issue.

FROGbots-4634, 2019-10-07:
@Matthew1231A Sorry to hear that! A couple questions:

 - Are you just using the unmodified webcam example program?
 - What phone are you using?
 - Are you using a powered USB hub?
 - What webcam are you using?
 - What version of the SDK are you using?
 - Does SDK's Vuforia webcam sample work?
 - What resolution are you trying to run the webcam at?
 - Does the EasyOpenCV internal camera example work?

NoahAndrews, 2019-10-07:
@Matthew1231A this may be your issue (workarounds are listed): https://ftcforum.usfirst.org/forum/ftc-technology/74722-bug-in-v5-2-that-breaks-support-for-web-cameras

FROGbots-4634, 2019-10-07:
@NoahAndrews yeah I was wondering the same, which is why I asked if the Vuforia webcam sample worked.

Matthew1231A, 2019-10-07:
@FROGbots-4634 the Vuforia camera sample works correctly **without** the workaround. I don't have the hardware with me, but I will try the workaround.

And to answer your questions:
We are using Moto G5's
Yes, we have a powered USB hub and yes, it has 5V at 2A
Logitech C270 webcam
FTC 5.2 SDK
Yes, the Vuforia SDK works
We are running at 320x240
The EasyOpenCV internal camera works

mtbbiker, 2019-10-07:
@FROGbots-4634 we have a similar issue, We use a C910 Webcam. The Vuforia example works (Actually we have used it successfully in Rover Ruckus). We would like to try OpenCV as alternative. Also we tried the internal camera examples and they work perfectly. Did we miss some configuration or calibration

FROGbots-4634, 2019-10-07:
@mtbbiker can you check and see if applying the workaround Noah mentioned makes a difference?

FROGbots-4634, 2019-10-08:
@Matthew1231A I just tested the EasyOpenCV webcam sample on a Moto G5 Plus running Android 7.0 with a Logitech C920 webcam without a powered USB hub using the stock SDK 5.2 and it ran flawlessly with and without the 64-bit workaround. (The Vuforia webcam sample also worked without the workaround which is a hrrmmmmm, but in any case, it worked).

haifengchicago, 2019-10-08:
Same issue here.  We have Moto G2 with Logitech C270 with a power USB hub.  Vuforia works, but the EasyOpenCV webcam is bluescreen.  I did the workaround, still no use.

NoahAndrews, 2019-10-08:
Can those of you having this problem try using the library with SDK 5.1?

FROGbots-4634, 2019-10-08:
Also, could someone please send a pre-built APK that's known to have the issue on your end?

NoahAndrews, 2019-10-08:
I can't reproduce on a Control Hub with the workaround in place

FROGbots-4634, 2019-10-08:
In addition to the "known bad" APK mentioned earlier, could someone please provide the logs from this encounter?

mtbbiker, 2019-10-08:
> @mtbbiker can you check and see if applying the workaround Noah mentioned makes a difference?

No Luck with the work around

NoahAndrews, 2019-10-08:
@mtbbiker Can you provide a log file and/or the known bad APK?

mtbbiker, 2019-10-08:
> @mtbbiker Can you provide a log file and/or the known bad APK?

I'll sent you the log, I am not too sure what you mean with the "bad" APK.

FROGbots-4634, 2019-10-08:
@mtbbiker "bad APK" as in an APK generated doing clicking Build --> Build Bundles / APK --> Build APK in Android Studio that you've verified has the issue.

mtbbiker, 2019-10-08:
@FROGbots-4634 a lengthy log, I can see a few errors (Hope this will provide some clue)
```
V/Robocol: received command: CMD_SET_MATCH_NUMBER(12607) 0
    received command: CMD_INIT_OP_MODE(12608) WebcamExample
I/RobotCore: ******************** START - OPMODE WebcamExample ********************
I/RobotCore: Attempting to switch to op mode WebcamExample
D/STMicroVL53L0X:: Checking to see if it's really a VL53L0X sensor...
D/STMicroVL53L0X:: Reg 0xC0 = ee (should be 0xEE)
D/STMicroVL53L0X:: Reg 0xC1 = aa (should be 0xAA)
    Reg 0xC2 = 10 (should be 0x10)
D/STMicroVL53L0X:: Reg 0x51 = 0 (should be 0x0099)
D/STMicroVL53L0X:: Reg 0x61 = 0 (should be 0x0000)
D/STMicroVL53L0X:: initial sig rate lim (MCPS) 0,250000
D/STMicroVL53L0X:: adjusted sig rate lim (MCPS) 0,250000
V/ThreadPool: container(0x0e7419b0: LinearOpMode) added id=2392 TID=0 count=1
V/RobotCore: thread: 'LinearOpMode main' starting...
D/CameraManager: doOpenCamera()...
V/Robocol: sending CMD_NOTIFY_INIT_OP_MODE(2855), attempt: 0
D/CameraManager: requesting permission for camera: Webcam(9828DE4F)
D/WebcamNameImpl: requesting permission for /dev/bus/usb/001/004
V/WebcamNameImpl: asyncRequestUsbPermission()...
D/WebcamNameImpl: permission already available for /dev/bus/usb/001/004
V/ThreadPool: container(0x05210305: default threadpool) added id=2393 TID=0 count=1
D/OpenCvViewport: CheckState(): user requested that we deactivate
    CheckState(): already deactivated
V/WebcamNameImpl: ...asyncRequestUsbPermission()
D/ContinuationSynchronizer: awaiting(camera open)...
D/WebcamNameImpl: permission for /dev/bus/usb/001/004=true
D/CameraManager: permission granted for camera: Webcam(9828DE4F)
    asyncOpenCamera(Webcam(9828DE4F))...
    asyncOpenWebcam(Webcam(9828DE4F))...
V/UsbResiliantWebcam: construct(0x0ef920dc)
D/UsbResiliantWebcam: openAssumingPermission()...
D/UvcContext: usb device is *not* UVC compatible, /dev/bus/usb/001/005
    found webcam: usbPath=/dev/bus/usb/001/004 vid=1133 pid=2093 serial=9828DE4F product=HD Pro Webcam C920
D/Uvc: [jni_context.cpp:195] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice()...
V/Uvc: API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice: enter...
D/libusb: [210.105822] [0000798d] libusb: debug [op_create] op_create: szUsbPath=/dev/bus/usb/001/004 szSysfsDir=1-1.2
D/Uvc: [device.cpp:613] uvc_create_uvc_device()...
    [libuvc/libuvc_internal.h:354] uvc_device::uvc_device()...
    [libuvc/libuvc_internal.h:383] acquired vid=1133 & pid=2093
    [libuvc/libuvc_internal.h:354] ...uvc_device::uvc_device()
    [device.cpp:613] ...uvc_create_uvc_device() rc=0
V/Uvc: API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice: ...exit
D/Uvc: [jni_context.cpp:195] ...Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice()
V/UvcDevice: construct(pointer=0x9bdfa200|null)
D/Uvc: [jni_device.cpp:108] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeGetLibUsbDevice()...
    [jni_device.cpp:108] ...Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeGetLibUsbDevice()
V/LibUsbDevice: construct(pointer=0x95f4fe00)
D/UvcDevice: open()...
D/Uvc: [jni_device.cpp:50] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeSetUsbDeviceInfo()...
    [jni_device.cpp:50] ...Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeSetUsbDeviceInfo()
    [jni_device.cpp:241] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeOpenDeviceHandle()...
V/Uvc: API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeOpenDeviceHandle: enter...
D/Uvc: [device.cpp:279] uvc_open()...
    [libuvc/libuvc_internal.h:364] uvc_device::open()...
D/libusb: [210.172326] [0000798d] libusb: debug [op_open] op_open: szUsbPath=/dev/bus/usb/001/004
D/Uvc: [libuvc/libuvc_internal.h:364] ...uvc_device::open() rc=0
    [device.cpp:295] calling libusb_reset_device()...
D/Uvc: [device.cpp:303] ...libusb_reset_device() called
    [device.cpp:311] libusb_open() = 0
    [libuvc/libuvc_internal.h:735] uvc_device_handle::uvc_device_handle()...
    [libuvc/libuvc_internal.h:735] ...uvc_device_handle::uvc_device_handle()
    [device.cpp:386] uvc_get_device_info()...
D/libusb: [210.591676] [0000798d] libusb: debug [op_get_config_descriptor] op_get_config_descriptor: config_index=0
    [210.591735] [0000798d] libusb: debug [op_get_config_descriptor] op_get_config_descriptor: config_index=0
D/Uvc: [device.cpp:986] uvc_scan_control()...
    [device.cpp:1041] uvc_parse_vc_header()...
    [device.cpp:1284] uvc_scan_streaming()...
    [device.cpp:1340] uvc_parse_vs_input_header()...
[... 506 lines cut ...]
D/UsbDeviceConnectionJNI: close
D/Uvc: [jni_device.cpp:126] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeReleaseRefDevice()...
V/Uvc: API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeReleaseRefDevice: enter...
D/Uvc: [device.cpp:882] uvc_unref_device()...
    [device.cpp:882] ...uvc_unref_device()
V/Uvc: API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeReleaseRefDevice: ...exit
[... cut ...]
V/Uvc: API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcFrameCallback_nativeReleaseCallbackState: enter...
    API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcFrameCallback_nativeReleaseCallbackState: ...exit
V/ThreadPool: container(0x05210305: default threadpool) removed id=2394 TID=31129 count=2
V/ThreadPool: container(0x05210305: default threadpool) removed id=2393 TID=31117 count=1
I/art: Background partial concurrent mark sweep GC freed 376757(13MB) AllocSpace objects, 116(2MB) LOS objects, 14% free, 91MB/107MB, paused 1.420ms total 121.869ms
```

FROGbots-4634, 2019-10-08:
@mtbbiker can you please confirm whether you have a C920 or a C910? You said that you had a C910, however according to the log the phone seems to think it's a C920.

FROGbots-4634, 2019-10-08:
@Matthew1231A @mtbbiker @haifengchicago the logs seem to indicate that a renderscript issue is the smoking gun. There was a change to the renderscript stuff in SDK v5.2. **Please test with SDK v5.1**

haifengchicago, 2019-10-09:
I just tested with SDK v5.1, still get a no image, just all black with the green rectangle in the middle.  The phone internal camera works with EasyOpenCV.  Vuforia also works with Webcam.  Moto G2, Logitech C270.

mtbbiker, 2019-10-09:
> @mtbbiker can you please confirm whether you have a C920 or a C910? You said that you had a C910, however according to the log the phone seems to think it's a C920.

@FROGbots-4634 my bad yes its the 920 (1080p) model.

haifengchicago, 2019-10-09:
Here is the log file from the SDK v5.1 run earlier.  Somewhere down it does say Render script failed.

10-08 22:11:27.267 12524 12722 I RobotCore: ******************** START - OPMODE WebcamExample ********************
10-08 22:11:27.268 12524 17442 I RobotCore: Attempting to switch to op mode WebcamExample
10-08 22:11:27.360 12524 17491 V RobotCore: thread: 'LinearOpMode main' starting...
10-08 22:11:27.377 12524 12794 V Robocol : sending CMD_NOTIFY_INIT_OP_MODE(23701), attempt: 0
10-08 22:11:27.428 12524 17491 D CameraManager: doOpenCamera()...
10-08 22:11:27.431 12524 17491 D CameraManager: requesting permission for camera: Webcam(D47A8AC0)
10-08 22:11:27.439 12524 17491 D WebcamNameImpl: requesting permission for /dev/bus/usb/001/004
10-08 22:11:27.439 12524 12524 D OpenCvViewport: CheckState(): user requested that we deactivate
10-08 22:11:27.439 12524 17491 V WebcamNameImpl: asyncRequestUsbPermission()...
10-08 22:11:27.439 12524 12524 D OpenCvViewport: CheckState(): already deactivated
10-08 22:11:27.442 12524 17491 D WebcamNameImpl: permission already available for /dev/bus/usb/001/004
10-08 22:11:27.443 12524 17491 V WebcamNameImpl: ...asyncRequestUsbPermission()
10-08 22:11:27.443 12524 17377 D WebcamNameImpl: permission for /dev/bus/usb/001/004=true
10-08 22:11:27.443 12524 17377 D CameraManager: permission granted for camera: Webcam(D47A8AC0)
10-08 22:11:27.443 12524 17491 D ContinuationSynchronizer: awaiting(camera open)...
10-08 22:11:27.447 12524 17377 D CameraManager: asyncOpenCamera(Webcam(D47A8AC0))...
10-08 22:11:27.447 12524 17377 D CameraManager: asyncOpenWebcam(Webcam(D47A8AC0))...
10-08 22:11:27.450 12524 17377 V UsbResiliantWebcam: construct(0x032919a3)
10-08 22:11:27.455 12524 17377 D UsbResiliantWebcam: openAssumingPermission()...
10-08 22:11:27.457 12524 17377 D UvcContext: found webcam: usbPath=/dev/bus/usb/001/004 vid=1133 pid=2085 serial=D47A8AC0 product=null
10-08 22:11:27.457 12524 17377 D Uvc     : [jni_context.cpp:195] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice()...
10-08 22:11:27.461 12524 17377 V Uvc     : API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice: enter...
10-08 22:11:27.465 12524 17377 D libusb  : [2526.403124] [000043e1] libusb: debug [op_create] op_create: szUsbPath=/dev/bus/usb/001/004 szSysfsDir=1-1.3
10-08 22:11:27.465 12524 17377 D libusb  : [2526.406548] [000043e1] libusb: debug [initialize_device] initialize_device(szUsbPath=/dev/bus/usb/001/004 szSysfsDir=1-1.3)
10-08 22:11:27.466 12524 17377 D libusb  : [2526.407164] [000043e1] libusb: debug [initialize_device] initialize_device(szUsbPath=(null) szSysfsDir=1-1)
10-08 22:11:27.466 12524 17377 E libusb  : [2526.407496] [000043e1] libusb: error [initialize_device] unable to open serial number sysfs; error ignored
10-08 22:11:27.466 12524 17377 D libusb  : [2526.407712] [000043e1] libusb: debug [initialize_device] initialize_device(szUsbPath=(null) szSysfsDir=usb1)
10-08 22:11:27.467 12524 17377 D Uvc     : [device.cpp:613] uvc_create_uvc_device()...
10-08 22:11:27.467 12524 17377 D Uvc     : [libuvc/libuvc_internal.h:354] uvc_device::uvc_device()...
10-08 22:11:27.467 12524 17377 D Uvc     : [libuvc/libuvc_internal.h:383] acquired vid=1133 & pid=2085
10-08 22:11:27.467 12524 17377 D Uvc     : [libuvc/libuvc_internal.h:354] ...uvc_device::uvc_device()
10-08 22:11:27.467 12524 17377 D Uvc     : [device.cpp:613] ...uvc_create_uvc_device() rc=0
10-08 22:11:27.467 12524 17377 V Uvc     : API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice: ...exit
10-08 22:11:27.467 12524 17377 D Uvc     : [jni_context.cpp:195] ...Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcContext_nativeCreateUvcDevice()
10-08 22:11:27.467 12524 17377 V UvcDevice: construct(pointer=0xb913fe38|null)
10-08 22:11:27.468 12524 17377 D Uvc     : [jni_device.cpp:108] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeGetLibUsbDevice()...
10-08 22:11:27.468 12524 17377 D Uvc     : [jni_device.cpp:108] ...Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeGetLibUsbDevice()
10-08 22:11:27.468 12524 17377 V LibUsbDevice: construct(pointer=0xb8f176f8)
10-08 22:11:27.468 12524 17377 D UvcContext: usb device is *not* UVC compatible, /dev/bus/usb/001/003
10-08 22:11:27.470 12524 17377 D UvcDevice: open()...
10-08 22:11:27.471 12524 17377 D Uvc     : [jni_device.cpp:50] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeSetUsbDeviceInfo()...
10-08 22:11:27.471 12524 17377 D Uvc     : [jni_device.cpp:50] ...Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeSetUsbDeviceInfo()
10-08 22:11:27.471 12524 17377 D Uvc     : [jni_device.cpp:241] Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeOpenDeviceHandle()...
10-08 22:11:27.471 12524 17377 V Uvc     : API: Java_org_firstinspires_ftc_robotcore_internal_camera_libuvc_nativeobject_UvcDevice_nativeOpenDeviceHandle: enter...
10-08 22:11:27.471 12524 17377 D Uvc     : [device.cpp:279] uvc_open()...
10-08 22:11:27.472 12524 17377 D Uvc     : [libuvc/libuvc_internal.h:364] uvc_device::open()...
[... cut ...]
10-08 22:11:29.363 12524 17505 E rsC++   : Couldn't initialize dispatch table
10-08 22:11:29.363 12524 17505 E UvcContext: [jni_context.cpp:141] renderScript initialization failed: err=0
10-08 22:11:29.363 12524 17505 E UvcFrame: [jni_frame.cpp:226] renderScript creation failed
10-08 22:11:29.414 12524 17505 D UvcContext: [jni_context.cpp:133] creating renderScript(/data/user/0/com.qualcomm.ftcrobotcontroller/cache/org.firstinspires.ftc.renderscript.cache,0x00000000,26)
10-08 22:11:29.414 12524 17505 E rsC++   : Couldn't initialize dispatch table

FROGbots-4634, 2019-10-09:
@haifengchicago what is the output if you run `adb shell getprop debug.rs.forcecompat`?

FROGbots-4634, 2019-10-10:
@Matthew1231A @haifengchicago @mtbbiker a HOTFIX 1.2 release has been pushed which works around an SDK bug that prevented the initialization of RenderScript on some devices, which resulted in failure to properly forward the webcam frames to the user pipeline. To upgrade, simply change the line in your gradle file to
```
implementation 'org.openftc:easyopencv:1.2'
```

haifengchicago, 2019-10-10:
@FROGbots-4634 Great, confirm that HOTFIX 1.2 addressed the issue.  You guys rocks!  This helps us a lot because this is much simpler than taking a frame from Vuforia (and with water mark).

closed 2019-10-10 by NoahAndrews

mtbbiker, 2019-10-10:
> @Matthew1231A @haifengchicago @mtbbiker a HOTFIX 1.2 release has been pushed which works around an SDK bug that prevented the initialization of RenderScript on some devices, which resulted in [...]

@FROGbots-4634 I can confirm that the work-around also fix our problem! Thank you for the help !