OpenFTC/EasyOpenCV #2: App crashes on initialization

anonamause10, 2019-10-13:
After installing EasyOpenCV, I tried to run it on my phone, but it crashes immediately after initialization. I am trying to run InternalCameraExample.java. When I check the Logcat, the error seems to be :
` java.lang.UnsatisfiedLinkError: No implementation found for long org.opencv.core.Mat.n_Mat(int, int, int) (tried Java_org_opencv_core_Mat_n_1Mat and Java_org_opencv_core_Mat_n_1Mat__III)
        at org.opencv.core.Mat.n_Mat(Native Method)
        at org.opencv.core.Mat.<init>(Mat.java:39)`

From this, I assume the problem is that the libOpenCvNative.so file is not being read by the robot controller for some reason. I have followed the instructions for the installation of EasyOpenCV as listed in the ReadME. The libOpenCvNative.so file is in the FIRST folder on the driver station.

Currently, our team is waiting to get new Moto E5's, so I am testing this using my personal Samsung galaxy s10e as the robot controller with a ZTE speed as the driver station. Thank you in advance for your help.

FROGbots-4634, 2019-10-13:
@anonamause10 I suspect the problem is that you are deploying a Robot Controller app with 64-bit native libraries, but `libOpenCvNative.so` is 32-bit. It appears that Android apps cannot mix and match 64-bit and 32-bit libraries. Please try applying [the workaround for webcam issues in v5.2.](https://github.com/FIRST-Tech-Challenge/SkyStone/wiki/Troubleshooting#problem-with-uvc-webcam-support-and-version-52-software-listed-092019). That will cause a 32-bit only Robot Controller app to be deployed.

FROGbots-4634, 2019-10-13:
also I just noticed you said *"The libOpenCvNative.so file is in the FIRST folder on the driver station."*. That file should be in the FIRST folder on the **robot controller**, not on the driver station.

anonamause10, 2019-10-13:
Yeah, I meant to say robot controller. I used the workaround, and the problem got fixed. Thank you for your help!

closed 2019-10-14 by FROGbots-4634