OpenFTC/EasyOpenCV #3: Camera does not support requested resolution

anonamause10, 2019-10-17:
I've been trying to get EasyOpenCV to work on my Webcam(Microsoft Lifecam HD 3000) but I keep getting an error saying that the resolution that I'm asking for does not work. My code is [here](https://github.com/anonamause10/6200-SkyStone-5.2/blob/c24472433ee18dee130c6a61074c1608b546a9f9/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/Camplz.java#L136), and I don't know what is wrong. According to [this](https://github.com/ftctechnh/ftc_app/blob/master/TeamCode/src/main/res/xml/teamwebcamcalibrations.xml) document, the resolution that I am using should be the correct one.

Thank you in advance for your help.

FROGbots-4634, 2019-10-17:
Does the Vuforia webcam sample work? Also which RC device are you using? And are you able to test with a different webcam model?

anonamause10, 2019-10-17:
The Vuforia webcam sample does work. The main problem here is with the webcam or USB hub, I think. Sometimes the webcam is detected with and address of 1118:2064, and it will work. Most of the time, the webcam is detected with an address of 0:0, where it doesn't work. I haven't figured out how to get the real address, but usually running the EasyOpenCV code, getting an error, and then restarting the robot gets the webcam to be properly detected.

FROGbots-4634, 2019-10-17:
Does the Vuforia sample work even when the serial shows as 0:0?

anonamause10, 2019-10-17:
It does run, but no display shows up on the RC phone. When the serial switches back to the real one, the Vuforia sample works fine.

FROGbots-4634, 2019-10-17:
Ok, well in that case it seems this is an SDK UVC driver issue or hardware issue.

closed 2019-10-17 by FROGbots-4634