OpenFTC/EasyOpenCV #4: Opencv internal camera can't  instantiated

ftc10015, 2019-12-29:
[image]

FROGbots-4634, 2019-12-30:
You're using EasyOpenCV 1.3 or higher, but still using sample code from <= 1.2.

1.3 had an API change which changes how cameras are instantiated.

closed 2019-12-30 by FROGbots-4634

ftc10015, 2019-12-30:
Thanks FROGbots, I changed to 1.2

FROGbots-4634, 2019-12-30:
That's a *valid* solution, although you lose access to the new features provided by 1.3..... all you need to do is change it to be:

```java
phoneCam = OpenCvCameraFactory.getInstance().createInternalCamera(OpenCvInternalCamera.CameraDirection.BACK, cameraMonitorViewId);
```

NoahAndrews, 2019-12-31:
In the future we should avoid having breaking changes in minor version updates.