OpenFTC/EasyOpenCV #8: Memory Leak during INIT phase

adrian27513, 2020-02-03:
So we have been having a memory leak issue for the past month during our qualifiers when we Initialize and let the program sit there for about 2 minutes. The exact error message we got is attached below. But today when we tried to recreate the error by letting it run in INIT continuously it took over an hour before it errored out. During competitions, we consistently got this error after 1-2 minutes every match even after restarting the phones before each one. When we asked around and searched online the only solution was to use the release function which we did.

The links to our autonomous codes are below:
https://github.com/Critical-Overload/CO_SkyStone_App/blob/master/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/AASkystoneSideBlue.java

https://github.com/Critical-Overload/CO_SkyStone_App/blob/master/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/AASkystoneSideRed.java

Thank you,
Adrian
Programmer for Critical Overload 12828

[image]

FROGbots-4634, 2020-02-03:
EasyOpenCV has been extensively optimised and profiled to ensure it does not leak memory. Therefore, this is almost certainly an issue with your code. A well-written pipeline will re-use previously allocated memory and never need to call release.

FROGbots-4634, 2020-02-05:
Looking briefly at your pipeline, it appears you're still using Mats *after you've released them* which can probably cause undefined behavior.... i.e. a "use after free" in C++

adrian27513, 2020-02-10:
I think our issue is solved now after rewriting the Pipeline. Thank you for your help.

closed 2020-02-10 by adrian27513