OpenFTC/EasyOpenCV #9: ERROR: package does not exist

AlexejheroYTB, 2020-02-04:
I am trying to install EasyOpenCV on a control hub. When I try to build it, the following error appears:

```
Building........................BUILD FAILED
org/firstinspires/ftc/teamcode/autonomous/RED.java(9:23): ERROR: package org.opencv.core does not exist
org/firstinspires/ftc/teamcode/autonomous/RED.java(14:26): ERROR: package org.opencv.imgproc does not exist
org/firstinspires/ftc/teamcode/autonomous/RED.java(15:30): ERROR: package org.openftc.easyopencv does not exist
```

I followed all of the instructions in the README file, and I did copy the `.so` file to the `FIRST` folder.

FROGbots-4634, 2020-02-04:
Did you perform a Gradle sync after editing the Gradle files? If so, did it succeed?

AlexejheroYTB, 2020-02-04:
I am using a robot controller.

AlexejheroYTB, 2020-02-04:
Oh, do I need to build the app with the edited gradle files on the robot controller?

AlexejheroYTB, 2020-02-04:
Wait, i'm dumb. I'm using a control hub with the ftc_http library, which makes it possible to upload code to the control hub directly from android studio. I did follow the instructions, but I added the EasyOpenCV library to Android Studio, and not the control hub

closed 2020-02-04 by AlexejheroYTB

FROGbots-4634, 2020-02-04:
Yeah, EasyOpenCV is not compatible OnBotJava (which is what ftc_http uses) without some extra work.

FROGbots-4634, 2020-02-04:
FYI, you can use ADB over network to upload the full app from Android Studio wirelessly, and using OpenRC-Turbo can make that process a nice bit faster than the stock SDK.

AlexejheroYTB, 2020-02-07:
> Yeah, EasyOpenCV is not compatible OnBotJava (which is what ftc_http uses) without some extra work.

How can I do that? I connected the control hub via ADB and built the app, but OnBotJava still says that there is a package missing.

FROGbots-4634, 2020-02-07:
I played around with this a while ago... IIRC after building the app with EasyOpenCV using Android Studio, the only other thing you had to do to make it work with OnBot was copy the sources JAR (of same version) into OnBot's lib folder.

**Please note that this is currently an unsupported way of using EasyOpenCV and I can't guarantee that you won't run into issues that you might not otherwise!**

Assuming you built the app from Android Studio using EasyOpenCV v1.3.2, then you will need to download [easyopencv-1.3.2-sources.jar](https://bintray.com/openftc/maven/download_file?file_path=org%2Fopenftc%2Feasyopencv%2F1.3.2%2Feasyopencv-1.3.2-sources.jar) and put it in the `FIRST/java/lib` folder on the internal storage of your Control Hub.

AlexejheroYTB, 2020-02-07:
I added the file to the control hub as you specified, but the same error still occurs.

AlexejheroYTB, 2020-02-07:
I installed the app on the control hub with the opmodes in it, and it worked with opencv. The reason why we use OnBotJava is because the build time is the smallest and the install time is 0. This is very helpful when making small changes to autonomous opmodes.

FROGbots-4634, 2020-02-07:
Just to make sure you put it in the right place, there should have been two JARs already in that folder, `android.jar` and `onbotjava-classes.jar` correct?

AlexejheroYTB, 2020-02-07:
Yep

FROGbots-4634, 2020-02-07:
Oh hmm I think I gave you the wrong file. I think it wants a compiled JAR not a sources JAR.

FROGbots-4634, 2020-02-07:
Ok, try deleting the file I gave you before, and putting this one in there instead: https://drive.google.com/open?id=1dScdAYt6e_EbG2B3anDjVdIK9bPHCEnY

AlexejheroYTB, 2020-02-07:
```
org/firstinspires/ftc/teamcode/autonomous/RED.java(9:23): ERROR: package org.opencv.core does not exist
org/firstinspires/ftc/teamcode/autonomous/RED.java(14:26): ERROR: package org.opencv.imgproc does not exist
```

But now there is no error for `org.opencv.easyopencv`

FROGbots-4634, 2020-02-07:
Oh right, how could I forget, you also need the JAR for plain old OpenCV too. Here you go: [opencv-repackaged-4.1.0-B-classes-compiled.jar](https://drive.google.com/open?id=1qOtzW2-a19HZQbqUM80rK1RWD6SfLTkS). (Just copy it into that same folder as well).

Aaaand this is one of the reasons why OnBot isn't officially supported, lol........

AlexejheroYTB, 2020-02-07:
Thank you very much!
It might be helpful for other people if there was a wiki for onbotjava installation instructions :)

Windwoes, 2021-09-20:
FYI, EOCV 1.5.0 now has official support for OnBotJava.