OpenFTC/EasyOpenCV #11: viewport container id

rohitdasari0424, 2020-02-22:
Hey, I tried your sample code and it's showing an error when I init saying that the viewport container that the user specified does not exist... How do I fix this problem?

FROGbots-4634, 2020-02-23:
Can you provide more details? What phone are you using? What SDK version? Have you modified the activity XML layout in any way?

rohitdasari0424, 2020-02-23:
I'm using the latest version of the FTC SDK with a motorola g5 as the robot controller and a samsung J327A as the driver controller. The specific error was: The view port container that the user specified does not exist. I did : int cameraMonitorViewId = hardwareMap.getAppContext.getResources().getIdentifier("cameraMonitorViewId","id",hardwareMap.AppContext.getPackageName()); and passed that as a parameter to the viewport id in the opencvinternalcamera. I have not modified the activity XML layout in any way.

rohitdasari0424, 2020-02-23:
[image]

FROGbots-4634, 2020-02-24:
Does the SDK's vuforia sample work?

FROGbots-4634, 2020-02-25:
Also if you could dump the main activity layout using the Layout Inspector that would be helpful too.

FROGbots-4634, 2020-03-05:
Are you still having this issue?

rohitdasari0424, 2020-03-05:
No I fixed it by downloading doge cv and it somehow ended up fixing the
problem

FROGbots-4634, 2020-03-05:
DogeCV is based off of EasyOpenCV so that doesn't entirely make sense, but ok...

closed 2020-03-05 by FROGbots-4634

rohitdasari0424, 2020-03-05:
Yeah, it really doesn’t make sense, but ig it worked and easyopencv is
awesome btw👌