OpenFTC/EasyOpenCV #12: OpenCvCameraException Error

ZMcElroy, 2020-03-10:
We are utilizing 2 Webcams and are using the sample code you provided for a single Webcam since we only use one of the webcams in each auto program. The very first time we try to run an auto program with the EasyOpenCV tied to it, it gives us the following error: 

Error: User code threw an uncaught exception: OpenCvCameraException - startStreaming() caled, but camera is not opened. 

I have attached a picture of the error as well. 
[image]

Once the error pops up, we can try to rerun the auto program and then it runs just fine. It just occurs the first time it is plugged into the robot and ran or when the robot is restarted. I looked online for the same error and noticed someone on Reddit had the same issue. I reached out to them and they suggested adding "Try Catch" statement which would return telemetry saying the camera hasn't started as opposed to the program crashing. We would then have to just stop and restart the program instead.

FROGbots-4634, 2020-03-10:
Could you provide a minimal log file from the Robot Controller which captures the error on the first run and the success on the subsequent run?

1. Close and swipe away the RC app
2. Use a file manager to delete `/sdcard/robotControllerLog.txt`
3. Start the RC app, and reproduce the error and then subsequent success
4. Close and swipe away the RC app
5. Reboot the phone to ensure that the filesystem has been synced to NAND (this wouldn't be a problem if the MTP protocol was less buggy....)
6. Copy the `/sdcard/robotControllerLog.txt` file to your computer and upload it here.

FROGbots-4634, 2020-03-10:
Also, do you experience this same behavior when running the SDK's Vuforia webcam sample?

ZMcElroy, 2020-03-10:
I went in and loaded the WebcamExample program to make sure it wasn't an issue with our program. I ran it and the same results happened. I did as you said to grab the log file and have attached it here. 

After clearing the file and resting the app here is what I did to reproduce the problem:
1. Selected the WebcamExample Program
2. Initialized it 
3. The error code came up
4. Went to Restart Robot on the drivers station and selected it
5. Initialized the program
6. Played the program and it worked fine. 

[robotControllerLog.txt](https://github.com/OpenFTC/EasyOpenCV/files/4312964/robotControllerLog.txt)

FROGbots-4634, 2020-03-10:
Well I see something very suspicious in your log...

```
11-27 22:09:37.786  8650  8934 D libusb  : [489.113857] [000022e6] libusb: debug [_originate_err] [linux_usbfs.c:798] originating err: -12(LIBUSB_ERROR_NOT_SUPPORTED)
11-27 22:09:37.786  8650  8935 V UvcDeviceHandle: construct(pointer=0x8d8a2420)
11-27 22:09:37.786  8650  8934 D Uvc     : [device.cpp:613] uvc_create_uvc_device()...
11-27 22:09:37.786  8650  8934 D Uvc     : [libuvc/libuvc_internal.h:354] uvc_device::uvc_device()...
11-27 22:09:37.786  8650  8934 D libusb  : [489.114131] [000022e6] libusb: debug [_originate_err] [descriptor.c:543] originating err: -5(LIBUSB_ERROR_NOT_FOUND)
........
11-27 22:09:37.789  8650  8934 D UvcContext: usb device is *not* UVC compatible, /dev/bus/usb/001/007
```

@NoahAndrews any thoughts?

ZMcElroy, 2020-03-10:
We are using Logitech C270 Webcams if you are referring to the UVC compatible portion.

ZMcElroy, 2020-03-10:
When I use the MultipleCameraExampleOpenCVAlongsideVuforia example program, I do not receive any errors at all.

ZMcElroy, 2020-03-10:
Here is the log file from when it worked with the Multiple Cameras and Vuforia/CV
[robotControllerLog.txt](https://github.com/OpenFTC/EasyOpenCV/files/4313443/robotControllerLog.txt)

FROGbots-4634, 2020-03-10:
Hmm interesting. I have a suspicion. Go back to the `WebcamExample` program. Find this part in the program:

```java
        /*
         * Open the connection to the camera device
         */
        webcam.openCameraDevice();
```

And replace it with:

```java
        /*
         * Open the connection to the camera device
         */
        webcam.openCameraDevice();
        sleep(1000);
        webcam.openCameraDevice();
```

And then test again and see what happens.

ZMcElroy, 2020-03-10:
Adding in what you suggested took care of the error message and that WebcamExample program works fine on the first try.

ZMcElroy, 2020-03-10:
I added that into one of our main pieces of code and it took care of the error message in there as well.

FROGbots-4634, 2020-03-10:
Really, very interesting. If you remove the sleep between calls to `openCameraDevice()`, does it still work?

ZMcElroy, 2020-03-10:
I checked in the WebcamExample code and yes, removing the sleep does allow it to work without any errors.

FROGbots-4634, 2020-03-10:
Hmm. I think what we're seeing here is that your particular phone takes longer to grant permission and open the camera than EasyOpenCV's timeout allows, which is 1 second.

The implementation of the webcam driver for Vuforia splits up the initialization into seperate steps for aquiring permission and opening, each with 1 second timeout. I could potentially switch to doing that, or I could keep it the current way and increase the timeout to 2 seconds... Hmmm...

What model RC phone are you using?

ZMcElroy, 2020-03-10:
I removed the sleep from our other main programs and it is still working in those as well.

ZMcElroy, 2020-03-10:
We are using Moto G5's

FROGbots-4634, 2020-03-10:
Interesting, a while ago I tested webcams on a Moto G5 Plus and it didn't run into any problems. If I were to generate an unofficial release or two, would you be intetested in testing?

ZMcElroy, 2020-03-10:
I don't mind pending they aren't too drastic of changes. We have our state championship this weekend so i'm a little hesitant to change too much. However, we don't leave until Friday.

FROGbots-4634, 2020-03-11:
Well since you have a workaround, why don't you keep that for now and we can revisit this later after your championship :)

FROGbots-4634, 2020-08-02:
Should be fixed in 1.4.0

closed 2020-08-02 by FROGbots-4634