OpenFTC/EasyOpenCV #16: Show error when open the app

adamrubinfeld, 2020-09-11:
i open the app and then the app show me that error
[image]

FROGbots-4634, 2020-09-11:
Please follow the directions included on the dialog. That will fix the problem :)

adamrubinfeld, 2020-09-12:
@FROGbots-4634 I have four build.common.gardle (teamCode, robotController, ftc-dashboard (of acme robotics) and for the all project) From which one I need to remove?

FROGbots-4634, 2020-09-12:
You might have multiple build.gradle files, but I doubt you have multiple build.common.gradle files. In any case, only one of them should have those entires so it should be fairly easy to determine which one you need to modify.

adamrubinfeld, 2020-09-12:
i removed from the public build.common.gardle and then that error print to the console

E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.qualcomm.ftcrobotcontroller, PID: 20412
    java.lang.UnsatisfiedLinkError: dlopen failed: file offset for the library "libVuforia.so" >= file size: 0 >= 0
        at java.lang.Runtime.loadLibrary0(Runtime.java:1016)
        at java.lang.System.loadLibrary(System.java:1657)
        at org.firstinspires.ftc.robotcore.internal.system.AppUtil.<clinit>(AppUtil.java:1692)
        at org.firstinspires.ftc.robotcore.internal.system.AppUtil.onApplicationStart(AppUtil.java:216)
        at org.firstinspires.ftc.robotcore.internal.system.RobotApplication.onCreate(RobotApplication.java:50)
        at android.app.Instrumentation.callApplicationOnCreate(Instrumentation.java:1125)
        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6062)
        at android.app.ActivityThread.-wrap1(Unknown Source:0)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1764)
        at android.os.Handler.dispatchMessage(Handler.java:105)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6944)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1374)

FROGbots-4634, 2020-09-13:
Well... that's interesting. It failed to load the _Vuforia_ library, which is not at all related to OpenCV. Just to confirm, you only removed [the two `arm64-v8a` entries from `build.common.gradle`](https://github.com/FIRST-Tech-Challenge/SkyStone/blob/master/build.common.gradle#L81) and nothing else?

FROGbots-4634, 2020-09-13:
Note: when I say "removed the two arm64-v8a entries", I mean changing the lines from

```
abiFilters "armeabi-v7a", "arm64-v8a"
```

to

```
abiFilters "armeabi-v7a"
```

closed 2020-09-14 by adamrubinfeld