OpenFTC/EasyOpenCV #17: Rev control hub - mtp mode

eyallampel1, 2020-09-20:
in your instructions 
connect Controller to your computer with a USB cable, put it into MTP mode, and drag 'n drop the file.

i cannot find a way to put the rev control into mtp mode .
no reference is found in the web

NoahAndrews, 2020-09-20:
The Control Hub is in MTP mode by default. It should show up in the Windows file explorer without you having to do anything special. On Mac, there's an application called Android File Transfer that you'll have to use.

eyallampel1, 2020-09-20:
yea i am connecting the rev control hub to the pc 
and i cannot see anything .
i will try with Android File Transfer maybe i will have more luck with that

FROGbots-4634, 2020-09-20:
Well, for Android handsets it is still necessary to put them into MTP. Maybe I can add a note in the readme that the CH enables it by default though.

closed 2020-09-20 by FROGbots-4634