OpenFTC/EasyOpenCV #18: External webcam issues

AlessioToniolo, 2020-09-23:
Hi I am a relatively new programmer in FTC and I am trying to implement EOCV for this year. We are running a control hub with an external webcam and most tutorials/code are for using the phone's camera. I checked through the javadocs and I am a little confused. I managed to type up some code for the initialization of the pipeline and webcam instances but it is not working. Is there compatibility with a external webcam? [Here is my code.](https://github.com/AlessioToniolo/FTC-EOCV-Ultimate-Goal/blob/master/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/vision/ultimategoalwebcampipeline.java) . Thank you!

FROGbots-4634, 2020-09-23:
Hi.

Please see the WebcamExample program: https://github.com/OpenFTC/EasyOpenCV/blob/master/examples/src/main/java/org/openftc/easyopencv/examples/WebcamExample.java

FROGbots-4634, 2020-09-25:
@AlessioToniolo Were you able to get it working?

AlessioToniolo, 2020-09-25:
Yes thank you I have not tested it on the robot but no compiler errors

closed 2020-09-25 by AlessioToniolo