OpenFTC/EasyOpenCV #26: Using Vuforia with OpenCV simultaneously

dgorbunov, 2021-01-17:
Is it possible to use Vuforia with OpenCV on one camera simultaneously or should I resort to switching between the two? It would be nice to be able to pass in the Vuforia frames into OpenCV like TensorFlow does, but it doesn't look possible. 

If not, then is it possible to make both a Vuforia and OpenCV instance off of one camera? I'm not sure...

Windwoes, 2021-01-17:
This is something that is not currently supported at the moment, but could be supported with a little bit of work. It brings numerous disadvantages, though, such as much higher overhead per frame as well as preventing you from being able to choose the specific resolution you want to run the camera at, and preventing you from having manual control over camera parameters such as exposure and ISO.

Running both a Vuforia and OpenCV instance off the same camera without EOCV support for pulling from Vuforia will definitely **not** work.

Windwoes, 2021-01-19:
I just released v1.4.4 which adds support for Vuforia passthrough mode. Take [OpenCvAndVuforiaOnSameCameraExample](https://github.com/OpenFTC/EasyOpenCV/blob/master/examples/src/main/java/org/openftc/easyopencv/examples/OpenCvAndVuforiaOnSameCameraExample.java) for a spin and let me know how it goes.

closed 2021-01-28 by Windwoes