OpenFTC/EasyOpenCV #27: Memory fault crashes Control Hub

dgorbunov, 2021-03-04:
When stopping an OpMode that uses Vuforia and OpenCV on the same camera the robot controller occassionally crashes.
This is the only part of the stack trace that throws an error:`A/libc: Fatal signal 11 (SIGSEGV), code 1, fault addr 0x28 in tid 2388 (RenderThread)`, but I've included the whole breakdown stack trace if it's related: https://pastebin.com/azhih4H8

Although our pipeline is leaking memory, this error happens intermittently and only started in the last week (our code has been leaking memory from the pipeline for ~2 months) so I'm not 100% sure memory leakage is related.

As far as I'm aware this is a camera related issue but it may be something deeper? An issue with the Control Hub? (See this [StackOverflow post](https://stackoverflow.com/questions/55795041/how-to-solve-android-fatal-signal-11-sigsegv-code-1-fault-addr-0x0-in-tid-xx))

Windwoes, 2021-03-05:
Unfortunately, the log you posted cuts off before showing the actual error. Also, are you using the viewport in GPU accelerated mode? If so, try switching to software rendering mode.

dgorbunov, 2021-03-05:
The issue seems to have gone away by not using Vuforia Passthrough mode. I'll try that also

closed 2021-03-15 by dgorbunov

Windwoes, 2021-03-15:
Did switching the rendering mode work?

dgorbunov, 2021-03-16:
No, it did not, but I'm no longer sure if the issue is with the camera. There is a chance that I may be having some other Hub issue. Nevertheless, I get a bunch of lines from the UVC library right before the crash, and it seems to be happening once every ~30 minutes. The entire app crashes and the RC closes its access point.
 
Here's another stack trace from a crash that doesn't even throw an error:
https://pastebin.com/RmcGx64B

Does everything look normal?

Windwoes, 2021-03-18:
Hmm, yeah I have also seen (exceedingly rare) random crashes of the Robot Controller when using webcams with EOCV on the Control Hub, even before I added the GPU-accelerated viewport mode or added any C++ code. I don't recall whether I've seen the same thing when just using Vuforia (no EOCV) or not.

dgorbunov, 2021-03-18:
Actually I stopped using Vuforia pass through mode completely. Hoping SDK v6.2 will fix this for me - they supposedly fixed a lot of bugs related to the RC crashing, or I may have some deeper hardware issue.