OpenFTC/EasyOpenCV #28: Robot controller crashes on initialization

AlexejheroYTB, 2021-06-03:
I'm trying to setup EasyOpenCV for FTC but I've ran into some issues. When pressing on the INIT button for an autonomous op mode which uses EasyOpenCV, the driver station stalls for a few seconds and then turns gray, as if it was disconnected from the control hub. Then, the control hub light starts blinking blue and the wifi network disappears. The only way to reconnect is to restart the control hub. (I'm using ftc_http to build to OnBotJava)

- [x] I've added the latest version of `libOpenCvNative.so` to `sdcard/FIRST` on the control hub
- [x] I'm using a compatible version of the robot controller app (6.1)
- [x] I've tried installing the latest official 32-bit version (5.2) as suggested in #2

We used EasyOpenCV last year but for some unknown reason it has stopped working since then.

AlexejheroYTB, 2021-06-03:
After further digging, I found this error in logcat:
```
FATAL EXCEPTION: opmode loop()
Process: com.qualcomm.ftcrobotcontroller
java.lang.NoClassDefFoundError: Failed resolution of: Lorg/openftc/easyopencv/OpenCvCameraFactory;
Caused by: java.lang.ClassNotFoundException: org.openftc.easyopencv.OpenCvCameraFactory
```

Windwoes, 2021-06-03:
I can try to take a look this weekend, but OBJ is not an officially supported use case :)

AlexejheroYTB, 2021-06-04:
I was able to resolve this issue by building and installing the robot controller app with easy open cv installed before attempting to run OBJ.

closed 2021-06-04 by AlexejheroYTB