OpenFTC/EasyOpenCV #29: unable to find libOpenCvNative.so when setting up/installing EasyOpenCV

robovines6955, 2021-09-16:
Following the link in the install instructions leads to a 404, and the file is not in the opencv repo

Windwoes, 2021-09-17:
Sorry, yeah, this is because I'm in the process of getting EOCV v1.5.0 released and it will require a new native lib file since I've upgraded the OpenCV core to v4.5.3.

Here's a link to the file (from repo history): https://github.com/OpenFTC/OpenCV-Repackaged/raw/566ee38f413609a95818f9fe566f4d90caf55e0e/doc/libOpenCvNative.so

robovines6955, 2021-09-17:
Sweet!! We were banging our heads pretty good trying to figure out what we were missing and figured we would just open an issue. Glad we weren't just missing it. Thanks!

closed 2021-09-17 by robovines6955

Windwoes, 2021-09-20:
FYI, v1.5.0 has been released

robovines6955, 2021-09-20:
Thanks mate, we will look at using the new version!