OpenFTC/EasyOpenCV #32: OnBotJava documentation not very clear.

BERENGENITA, 2021-09-29:
When using EasyOpenCV with OnBotJava, how should the .aar file be uploaded to OnBotJava? In the same way .java classes are uploaded or in a different way?

As a new FTC competitor, I find the instructions unclear and kindly ask the maintainers to improve them.

BERENGENITA, 2021-09-29:
I have also read that the fact that OnBotJava does not support .aar files is a known limitation per the release information available on the [README](https://github.com/FIRST-Tech-Challenge/FtcRobotController/tree/v7.0):
`Known limitation - .aar files with assets are not supported.`

Windwoes, 2021-09-29:
> When using EasyOpenCV with OnBotJava, how should the .aar file be uploaded to OnBotJava? In the same way .java classes are uploaded or in a different way?

Yes, uploaded from that same menu.

> Known limitation - .aar files with assets are not supported

This is not an issue, EOCV does not rely on assets.

>  I find the instructions unclear and kindly ask the maintainers to improve them.

Noted :)

BERENGENITA, 2021-09-29:
Thank you for the reply.

One quick question, what does `dexification` mean, as mentioned in the instructions? I could not find it online.

Windwoes, 2021-09-29:
When you upload the AAR you will get a dialog box saying that the library is being processed for FTC. Behind the scenes it is DEXing it. Android runs DEX bytecode, but JARs/AARs contain JVM bytecode. DEXing is converting the JVM bytecode to DEX bytecode.

closed 2021-09-29 by BERENGENITA