OpenFTC/EasyOpenCV #37: Modifying the EasyOpenCV library errors

BERENGENITA, 2021-11-11:
I'm trying to modify and recompile the EasyOpenCV library. I've already cloned the repo and got gradle to behave. However, it seems like the JNI is not being recognized properly. 

I see the following error `Class 'OpenCvSwitchableWebcamImpl' must either be declared abstract or implement abstract method 'shouldTriggerWarningSound()' in 'GlobalWarningSource'`

Also, in the C++, OpenCV doesn't seem to be recognized. For example, in `vuforia_passthrough_jni.cpp`, I have the error `'opencv2/core.hpp' file not found`.

Windwoes, 2021-11-11:
Regarding the first error, it sounds like you are trying to build against the SDK 7.0 RobotCore without my recent commit to fix compatibility.

As for the second, it sounds like you haven't cloned OpenCV-Repackaged into the same parent directory.

Windwoes, 2021-11-14:
Any luck?

BERENGENITA, 2021-11-15:
When you say same parent directory, do you mean that if my EasyOpenCV clone is in the Desktop, I should clone the OpenCV-Repackaged also in the Desktop?

Windwoes, 2021-11-15:
That's correct :)

Windwoes, 2021-11-15:
Alternatively, you can modify the CMake script: https://github.com/OpenFTC/EasyOpenCV/blob/master/easyopencv/CMakeLists.txt#L27

closed 2021-11-17 by BERENGENITA