OpenFTC/EasyOpenCV #38: Segmentation fault in recording pipeline

JJTech0130, 2021-11-23:
Log:
```
I/RobotCore: ******************** START - OPMODE TeamElementDemo ********************
I/RobotCore: Attempting to switch to op mode TeamElementDemo
V/ThreadPool: container(0x0c3ec91a: LinearOpMode) added id=157 TID=0 count=1
V/RobotCore: thread: 'LinearOpMode main' starting...
I/System.out: OpenCvCameraBase ctor: EasyOpenCV version 1.5.1
I/System.out: startRecordingPipeline()
V/Robocol: sending CMD_NOTIFY_INIT_OP_MODE(186), attempt: 0
V/Robocol: sending CMD_STREAM_CHANGE(187), attempt: 0
W/google-breakpad: ### ### ### ### ### ### ### ### ### ### ### ### ###
    Chrome build fingerprint:
    7.0
    42
    ### ### ### ### ### ### ### ### ### ### ### ### ###
A/libc: Fatal signal 11 (SIGSEGV), code 1, fault addr 0x1c in tid 4516 (LinearOpMode ma)
```
Code:
```java
OpenCvWebcam webcam = OpenCvCameraFactory.getInstance().createWebcam(hardwareMap.get(WebcamName.class, "Webcam 1"));

// Record for easier debugging
webcam.startRecordingPipeline(
        new PipelineRecordingParameters.Builder()
                .setBitrate(4, PipelineRecordingParameters.BitrateUnits.Mbps)
                .setEncoder(PipelineRecordingParameters.Encoder.H264)
                .setOutputFormat(PipelineRecordingParameters.OutputFormat.MPEG_4)
                .setFrameRate(30)
                .setPath("/Pictures/pipeline_rec.mp4")
                .build());
```

Windwoes, 2021-11-24:
What device are you running on? Also, you need to specify the path with like /sdcard/ or use the API to get the external storage path.

JJTech0130, 2021-11-24:
Yeah, I changed it to /sdcard but forgot to update the issue. It's running on a control hub.

Windwoes, 2021-11-27:
So, I never actually tested the recording feature on a control hub... I do know it worked on Nexus 5 phones.

Windwoes, 2021-12-09:
I can reproduce this not working on a Control Hub, but I don't get a segfault. Instead I get a failure in the RockChip video codec initialization: 

```java
1969-12-31 19:04:33.937 201-793/? E/ROCKCHIP_LOG: Rockchip_OSAL_SharedMemory_Open(391): 111ion_client_create Error: 7
1969-12-31 19:04:33.938 201-793/? E/ROCKCHIP_RM: Rockchip_OMX_Check_Resource(282): pRockchipComponent->codecType 2, Line:282
1969-12-31 19:04:33.941 201-907/? E/OMXNodeInstance: getParameter(c9002f:rk._encoder.avc, ParamConsumerUsageBits(0x6f800004)) ERROR: UnsupportedIndex(0x8000101a)
1969-12-31 19:04:33.951 201-631/? E/OMXNodeInstance: setConfig(c9002f:rk._encoder.avc, ConfigPriority(0x6f800002)) ERROR: UnsupportedIndex(0x8000101a)
1969-12-31 19:04:33.952 201-201/? E/OMXNodeInstance: setConfig(c9002f:rk._encoder.avc, ConfigOperatingRate(0x6f800003)) ERROR: UnsupportedIndex(0x8000101a)
1969-12-31 19:04:33.954 201-632/? E/OMXNodeInstance: getConfig(c9002f:rk._encoder.avc, ConfigAndroidIntraRefresh(0x6f60000a)) ERROR: UnsupportedIndex(0x8000101a)
```

I made a standalone test app to test using the MediaRecorder with an external surface (what EOCV does) as well as binding directly to an Android `Camera` (v1) instance, and both of those work fine on a Nexus 5 and fail with the above error on a Control Hub. I think something is fundamentally broken with the video encoder engine. Surprisingly, the Open Camera app is able to record video on the Control Hub, so that indicates that maybe there are certain configurations that work and others that are broken, I'm not sure. @NoahAndrews any thoughts?