OpenFTC/EasyOpenCV #39: 32 bit or 64 bit

shelbyRobots, 2021-11-28:
The instructions specify use of the 32 bit armeabi-v7a .so file, but OpenCV-Repackaged has both versions in docs/native-libs, both files are available.  We are shifting to eocv from an old hard-to-maintain pre-eocv home grown vuforia/opencv/camera loading manager.
In our existing stuff, we had built and were using opencv as 64 bit.
Is it possible to use the 64 bit arm64-v8a .so?
Thanks

Windwoes, 2021-11-28:
The 64-bit version is there because for the OBJ bundle, I need to support 64-bit prebuilt apps.

Currently OpenCV-Repackaged only supports dynamically loading the 32-bit version. As far as I'm aware, the 64-bit version provides no performance advantage. I would be happy to be proved wrong on that with some benchmarks, though.

shelbyRobots, 2021-11-29:
I will try to get some metrics when I can on comparison of 32 vs 64 bit.
I also wondered why conversion to Mat was done like it is.  Would Util bitmatToMat (in opencv repackaged) not work?

Windwoes, 2021-11-29:
> I also wondered why conversion to Mat was done like it is. Would Util bitmatToMat (in opencv repackaged) not work?

Using the SDK's built-in function for getting a bitmap from the UVC frame and then using bitmatToMat is less efficient (by a measurable amount) and it also suffers from compatibility issues due to renderscript usage, see issue #1 

> Is it easy/possible/working to build OpenCv-Repackaged locally with the modified NATIVE_LIB_MD5 set to the 64 bit .so in DynamicOpenCvNativeLibLoader.java?

Sure, you could do that if you wanted.

alan412, 2022-04-29:
I would also like to see a 64-bit version so students won't have to modify build.common.gradle

Windwoes, 2022-04-30:
Okay so the thing about supporting 64-bit with dynamic loading is that it would mean requiring users to copy both the 64 **and** 32bit native libraries to the internal storage and then add code to check for both, copy both to the protected storage etc.... which I'm not really sure that I'm in love with.

What I could do perhaps is create a variant of the artifact like the bundle for OnBotJava where both the 64-bit and 32-bit libraries are bundled inside the artifact. It would allow installing EOCV with only the implementation gradle line, at the expense of adding probably around 15MB to the APK...

NoahAndrews, 2022-04-30:
What if you distributed the native libraries as a single zip file that EasyOpenCV would extract? That would keep the set up process just as simple as it is now.

Windwoes, 2022-04-30:
Hmm that's an interesting idea.

seadragonwang, 2022-11-11:
I have a related issues, Before I remove arm64-v8a, my WebCamera can capture images, but after I removed arm64-v8a, pictures captured are all black, can you help me on this?

Windwoes, 2022-11-11:
EOCV will not run with 64-bit build enabled so I'm not sure how you managed to get images out of it before disabling.

Windwoes, 2022-12-22:
Starting with EOCV v1.5.3 I've made the decision to bundle the native libraries by default, including 64-bit.

closed 2022-12-22 by Windwoes