OpenFTC/EasyOpenCV #40: Loading OpenCV error

Eeshwar-Krishnan, 2022-02-22:
We just encountered an interesting issue with EasyOpenCV. After redownloading from Android Studio, we get the "An error occurred while loading the OpenCV native library. Any OpenCV-enabled opmodes will crash" error on the driver hub. Following this, the app will immediately restart and then EOCV will proceed to work fine. It happens every time we download (which has been slowing down development a lot since we need to wait for two app restarts essentially). The correct .so file is in the FIRST folder, and I verified EOCV does in fact work after the second restart. I can try to get an error log, but I don't see anything noteworthy when it crashes. I don't know if this is related, but we used the EOCV Apriltag plugin, and most of the time when the app restarts and that error isn't thrown, the camera doesn't open correctly and the apriltag pipeline freezes. This is all on v1.5.1

Windwoes, 2022-02-22:
Interesting. Without logs, it's hard to really make any intelligent comment.

Eeshwar-Krishnan, 2022-02-23:
Managed to recreate the issue, interestingly by redownloading libOpenCv it is now occuring once every few downloads instead of every time. Didn't manage to grab logs, for some reason logcat is empty? I will attempt again tonight. The one time I caught something it looked like an unsatisfied link exception

Windwoes, 2022-02-23:
What do you mean logcat is empty?
The logcat pane in AS will clear when the app restarts, but you should be able to view the entire logcat with "adb logcat".

Windwoes, 2022-02-28:
Well, the cause of the OpenCV loading error is..... interesting....

```
E/OpenFTC-OpenCV-Repackaged-Loader: Error occurred while loading OpenCV native library! Any OpenCV-enabled OpModes will crash
E/OpenFTC-OpenCV-Repackaged-Loader: java.lang.UnsatisfiedLinkError: dlopen failed: library "libc++_shared.so" not found
```
I do not see any logical reason why it can't find `libc++_shared.so` - that is actually bundled inside the OpenCV-Repackaged AAR in the "normal" place for native libraries.

What's even more perplexing is that it also fails to load the SDK's own native library:

```
com.qualcomm.ftcrobotcontroller E/ThreadPool: java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/data/app/com.qualcomm.ftcrobotcontroller-1/base.apk"],nativeLibraryDirectories=[/data/app/com.qualcomm.ftcrobotcontroller-1/lib/arm, /data/app/com.qualcomm.ftcrobotcontroller-1/base.apk!/lib/armeabi-v7a, /system/lib, /vendor/lib]]] couldn't find "libRobotCore.so"
```

This seems to indicate that it's not an issue with OpenCV specifically.

Moreover, before either of the native library failures, there's this:

```
E/AndroidRuntime: FATAL EXCEPTION: SoundPlayer-#0
    Process: com.qualcomm.ftcrobotcontroller, PID: 3245
    android.content.res.Resources$NotFoundException: File res/raw/ss_r2d2_up.wav from drawable resource ID #0x7f0d0019
        at android.content.res.ResourcesImpl.openRawResourceFd(ResourcesImpl.java:291)
        at android.content.res.Resources.openRawResourceFd(Resources.java:1267)
        at com.qualcomm.ftccommon.SoundPlayer.getMsDuration(SoundPlayer.java:659)
        at com.qualcomm.ftccommon.SoundPlayer.ensureLoaded(SoundPlayer.java:476)
        at com.qualcomm.ftccommon.SoundPlayer.loadAndStartPlaying(SoundPlayer.java:447)
        at com.qualcomm.ftccommon.SoundPlayer$3.run(SoundPlayer.java:306)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
        at com.qualcomm.robotcore.util.ThreadPool$ThreadFactoryImpl$1.run(ThreadPool.java:793)
        at java.lang.Thread.run(Thread.java:761)
     Caused by: java.io.FileNotFoundException: This file can not be opened as a file descriptor; it is probably compressed
```

It looks like the sound player is causing a fatal exception that ends up shutting down the ART VM, and maybe the native library loads fail because the ART VM is shutting down? cc @NoahAndrews

Windwoes, 2022-02-28:
Are you using the stock SDK or are you using OpenRC Turbo?

Eeshwar-Krishnan, 2022-02-28:
Stock SDK, latest version

Windwoes, 2022-02-28:
Can you go to Build -> Build APK in Android Studio and upload the built APK to Google Drive?

Windwoes, 2022-03-01:
Also, please try commenting out any OpenCV-related Gradle dependencies, and temporarily removing any code that uses OpenCV, and see if you experience issues still. Judging by the fact that the SDK's native library fails to load, I expect that you'd also see failure to communicate with the embedded Expansion Hub in the Control Hub.

Windwoes, 2022-03-01:
I tested building an APK and then going in and deleting the sound file in question from the /res/raw folder and then trying to install that APK in an attempt to reproduce what you saw - and the APK failed to install with the following error:

```
adb: failed to install TeamCode-debug.apk: Failure [INSTALL_PARSE_FAILED_NO_CERTIFICATES: Failed to collect certificates from /data/app/vmdl71454183.tmp/base.apk: /data/app/vmdl71454183.tmp/base.apk: File res/raw/ss_r2d2_up.wav in manifest does not exist]
```

It seems like something is really really screwy with your setup and I'm not sure what. But it definitely does not look like an OpenCV issue.

Windwoes, 2022-03-01:
If you have access to a Linux machine can you try running the following on the APK and paste the output here:

```
unzip -lv TeamCode-debug.apk | grep wav
```

Windwoes, 2022-03-14:
Any update on this?