OpenFTC/EasyOpenCV #42: java.lang.ClassNotFoundException: com.qualcomm.robotcore.wifi.WifiDirectAssistant$1

AlexejheroYTB, 2022-02-23:
Hello, I haven't been here in a while.

I'm installing EasyOpenCV on SDK 1.7.1 with a fully up-to-date Control Hub.
I added the external library through OnBotJava, as the guide said.
Building works both via the web and via ftc_http, but when running the code I get the following exception, and the robot controller crashes:

> 02-23 14:11:27.702  8793  8793 E ClassManager: java.lang.ClassNotFoundException: com.qualcomm.robotcore.wifi.WifiDirectAssistant$1
02-23 14:11:27.703  8793  8793 E ClassManager:  at java.lang.Class.classForName(Native Method)
02-23 14:11:27.705  8793  8793 E ClassManager:  at java.lang.Class.forName(Class.java:400)
02-23 14:11:27.707  8793  8793 E ClassManager:  at org.firstinspires.ftc.robotcore.internal.opmode.ClassManager.classNamesToClasses(ClassManager.java:214)
02-23 14:11:27.709  8793  8793 E ClassManager:  at org.firstinspires.ftc.robotcore.internal.opmode.ClassManager.processAllClasses(ClassManager.java:257)
02-23 14:11:27.710  8793  8793 E ClassManager:  at com.qualcomm.ftccommon.ClassManagerFactory.processAllClasses(ClassManagerFactory.java:94)
02-23 14:11:27.711  8793  8793 E ClassManager:  at org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity.onCreate(FtcRobotControllerActivity.java:356)
02-23 14:11:27.711  8793  8793 E ClassManager:  at android.app.Activity.performCreate(Activity.java:6709)
02-23 14:11:27.712  8793  8793 E ClassManager:  at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1118)
02-23 14:11:27.712  8793  8793 E ClassManager:  at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2619)
02-23 14:11:27.712  8793  8793 E ClassManager:  at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2727)
02-23 14:11:27.712  8793  8793 E ClassManager:  at android.app.ActivityThread.-wrap12(ActivityThread.java)
02-23 14:11:27.713  8793  8793 E ClassManager:  at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1478)
02-23 14:11:27.713  8793  8793 E ClassManager:  at android.os.Handler.dispatchMessage(Handler.java:102)
02-23 14:11:27.713  8793  8793 E ClassManager:  at android.os.Looper.loop(Looper.java:154)
02-23 14:11:27.714  8793  8793 E ClassManager:  at android.app.ActivityThread.main(ActivityThread.java:6121)
02-23 14:11:27.714  8793  8793 E ClassManager:  at java.lang.reflect.Method.invoke(Native Method)
02-23 14:11:27.714  8793  8793 E ClassManager:  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:905)
02-23 14:11:27.715  8793  8793 E ClassManager:  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:795)

The error is apparently thrown by the `createWebcam` function.

```cs
OpenCvCameraFactory factory = OpenCvCameraFactory.getInstance();
WebcamName camera = hardwareMap.get(WebcamName.class, "Webcam 1");
OpenCvCamera ocvCam = factory.createWebcam(camera);
```

AlexejheroYTB, 2022-02-23:
I was mistaken, the exception reported above might have been an induced error.
I've located another exception which happened before the first one which makes a bit more sense.

```
java.lang.NoSuchMethodError: No virtual method createWebcam(Lorg/firstinspires/ftc/robotcore/external/hardware/camera/WebcamName;)Lorg/openftc/easyopencv/OpenCvCamera; in class Lorg/openftc/easyopencv/OpenCvCameraFactory; or its super classes (declaration of 'org.openftc.easyopencv.OpenCvCameraFactory' appears in /storage/emulated/0/FIRST/ExternalLibraries/EasyOpenCV_v1_5_1_OBJ_bundle_aar/dex/classes.dex)
   at org.firstinspires.ftc.teamcode.auto.MainAuto.opmode_run(MainAuto.java:43)
   at org.firstinspires.ftc.teamcode.common.opmodes.HtechLinearOpMode$LinearOpModeHelper$1.run(HtechLinearOpMode.java:263)
   at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:737)
   at org.firstinspires.ftc.teamcode.common.opmodes.HtechLinearOpMode$LinearOpModeHelper.run(HtechLinearOpMode.java:258)
   at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
   at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
   at com.qualcomm.robotcore.util.ThreadPool$ThreadFactoryImpl$1.run(ThreadPool.java:793)
   at java.lang.Thread.run(Thread.java:761)
```

Windwoes, 2022-02-23:
I can try to reproduce this later, but this kind of seems like something is messed up due to how you set up EOCV in OBJ before I provided the official OBJ bundle. I might suggest backing up your code, nuking all the OnBotJava files on the Control Hub storage, and then reloading your code and the stock 7.1 APK.

AlexejheroYTB, 2022-02-23:
I will attempt that tomorrow. Thank you for the suggestion.

AlexejheroYTB, 2022-02-24:
This seemed to do the trick. I deleted the FIRST folder from the sdcard files of the Control Hub and the issue was resolved. Thank you for your support (again)!

closed 2022-02-24 by AlexejheroYTB

Windwoes, 2022-02-24:
Glad it's working :)