OpenFTC/EasyOpenCV #44: EasyOpenCV causes app error in Control Hub

tom-ricci, 2022-04-25:
In the newest version of the Control Hub's firmware, the app does not like to load when using EasyOpenCV, and seems to end up making either the Control Hub itself or the app bootloop, I'm not completely sure which. It's caused by removing `"arm64-v8a"` in the Gradle config.

Windwoes, 2022-04-25:
Would you be able to grab logs from the encounter?

tom-ricci, 2022-04-25:
Nothing is logged via ADB (as far as I remember, I can check soon); are there any other logs that would be helpful?

Windwoes, 2022-04-25:
If the app goes down, there definitely should be something printed to ADB logcat. Since Android Studio clears the logcat pane when relaunching the app, you may need to run `adb logcat` from a separate terminal window to capture the crash.

tom-ricci, 2022-04-28:
Okay, I'll let you know the results the next time I can test it out.

Windwoes, 2022-05-07:
Any update on this?

tom-ricci, 2022-05-07:
Sorry, I don't have anything at the moment

tom-ricci, 2022-05-10:
I got my hands on a control hub again (and the two others I've been having trouble with) and it just worked? I have no idea what happened between then and now, but for now it seems to be fine. I'll reopen the issue again with some more logs if this happens again.

closed 2022-05-10 by tom-ricci