OpenFTC/EasyOpenCV #47: Broken by change in FTC SDK 8.0

alan412, 2022-09-13:
The release notes of https://github.com/FIRST-Tech-Challenge/FtcRobotController for FTC SDK 8.0 say they break EasyOpenCV.

Windwoes, 2022-09-14:
Yeah. The fix is one line though I'm going to try to have it out tonight.

alan412, 2022-09-14:
That would be fantastic.  We are putting on a robot in 3 weeks scrimmage and it would be good to have everyone on 8.0

MDNich, 2022-09-14:
Yes!! Thank you so much @Windwoes. All of your hard work in EasyOpenCV is much appreciated and heavily used by my team (10738 Bears). Thank you so much for all you do. It probably goes unsung, but your work is truly helpful to almost all FTC teams with a webcam. Thanks!!!

lukasmwerner, 2022-09-14:
@Windwoes I just put together a PR so you can merge remote. I didn't increment the build.gradle but I can add that really quick if you want

Windwoes, 2022-09-14:
Well, theoretically I have artifacts for 1.5.2 built but I can't test right now because ADB is freaking out and the server is crash looping. I'm going to pick this up tomorrow given that it's midnight and I have a 7:45 class tomorrow :sob:

MDNich, 2022-09-14:
When you have a moment @Windwoes could you upload that artifact here? If possible the .so library for Control hub. Thank you so much.

Windwoes, 2022-09-14:
v1.5.2 release is now live, artifacts should be available on Maven Central within 30 mins (or whenever their sync job decides to run).

closed 2022-09-15 by Windwoes