OpenFTC/EasyOpenCV #50: OpMode crashes on initialization when creating `OpenCvCamera`

dheerajpv, 2022-10-15:
I currently have code like this that runs on initialization: 
```java
int cameraMonitorViewId = hardwareMap.appContext.getResources().getIdentifier("cameraMonitorViewId", "id", hardwareMap.appContext.getPackageName()); // code makes it up to here
WebcamName webcamName = hardwareMap.get(WebcamName.class, RobotMap.Vision.WEBCAM); // crashes on this line
OpenCvCamera camera = OpenCvCameraFactory.getInstance().createWebcam(webcamName, cameraMonitorViewId);
camera.openCameraDeviceAsync(new OpenCvCamera.AsyncCameraOpenListener() {
    @Override
    public void onOpened() {
        camera.startStreaming(1280, 720, OpenCvCameraRotation.UPRIGHT);
        camera.setViewportRenderer(OpenCvCamera.ViewportRenderer.GPU_ACCELERATED);
        camera.setPipeline(new SimplePipeline());
    }
    @Override
    public void onError(int errorCode) {}
});
```

However, this code keeps crashing the OpMode with no error reported on the driver station. There are some errors reported in logcat:
```
java.lang.NoClassDefFoundError: Failed resolution of: Lorg/firstinspires/ftc/robotcore/internal/opmode/OpModeManagerImpl;
	at org.openftc.easyopencv.LIFO_OpModeCallbackDelegate.getInstance(LIFO_OpModeCallbackDelegate.java:42)
	at org.openftc.easyopencv.OpenCvCameraBase.<init>(OpenCvCameraBase.java:110)
	at org.openftc.easyopencv.OpenCvWebcamImpl.<init>(OpenCvWebcamImpl.java:102)
	at org.openftc.easyopencv.OpenCvCameraFactoryImpl.createWebcam(OpenCvCameraFactoryImpl.java:101)
	at org.firstinspires.ftc.teamcode.Robot.initCamera(Robot.java:76)
	...
Caused by: java.lang.ClassNotFoundException: Didn't find class "org.firstinspires.ftc.robotcore.internal.opmode.OpModeManagerImpl" on path: DexPathList[[zip file "/data/app/com.qualcomm.ftcrobotcontroller-2/base.apk"],nativeLibraryDirectories=[/data/app/com.qualcomm.ftcrobotcontroller-2/lib/arm, /data/app/com.qualcomm.ftcrobotcontroller-2/base.apk!/lib/armeabi-v7a, /system/lib, /vendor/lib]]
	at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:380)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
        ... 18 more
```

I'm using FTCLib, so I followed the instructions [here](https://docs.ftclib.org/ftclib/installation) to install FTCLib and EOCV. Any help is appreciated.

Windwoes, 2022-10-15:
This is almost certainly due to FTC Lib not using the latest version of EOCV.

SDK v8.0 broke compatibility with EOCV; this breakage was fixed in v1.5.2.

Windwoes, 2022-10-15:
I messaged the FTC Lib maintainer on Discord; he will be trying to update FTCLib this weekend.

Closing this since it is not an issue with EOCV :')

closed 2022-10-15 by Windwoes