OpenFTC/EasyOpenCV #53: Camera Stream Display Rotated 90° to the Left

baqwas, 2022-11-11:
Hello!

First of all many, many thanks for the solution!

My Camera Stream image is _rotated_ to the left by 90 degrees. I have not changed (not that ambitious) any statement in the code from this repository except to envelope it into an Autonomous OpMode that works with the default TFOD signal sleeve detection where the Camera Stream is upright. Similarly, I dropped in the AprilTags example in the same Autonomous OpMode and the Camera Stream is upright.

Is there any _easy_ fix to ensuring that the default camera view is upright when using the code from this repository? (As you can well imagine, I would prefer to stay within _my job grade_ and not get into Easy Mat operations with the frames!). Thanks again. The entire community (those who want the extra 10 points) owe you a big one. Don't hesitate to let us know how we send some recognition (tangible or intangible) your way.

Kind regards.

baqwas, 2022-11-11:
Sorry for posting yet again in haste!
Found the solution:
OpenCvCameraRotation.**UPRIGHT**

Kind regards.

closed 2022-11-11 by baqwas

Windwoes, 2022-11-11:
The default rotation for the webcam example is in fact upright so I'm not sure how yours was set to something else.

baqwas, 2022-11-11:
Mystery solved!

Yet another user/operator _self-inflicted_ issue: I mixed up code from two repositories - OpenFTC and KookyBotz!

Regards.