OpenFTC/EasyOpenCV #55: Code Crashes Upon Run on FTC Control Hub

violet-flowers, 2022-11-17:
My team is trying to use EasyOpenCV to detect the different images on our signal sleeve cone. We had the example up and running ~6 weeks ago but have had trouble getting it working again. Every time we try and run an EasyOpenCV enabled op mode the Control Hub crashes and throws the exception "FATAL EXCEPTION: LinearOpMode main Process: UnsatisfiedLinkError: dlopen failed: library "libopencv_java4.so" not found. " We moved the library into our FIRST Folder on the internal storage, so I am unsure as to why we still receive this exception.

Windwoes, 2022-11-17:
Can you provide the robot controller logs?

Windwoes, 2022-12-03:
@violet-flowers Are you by chance using EOCV with OnBotJava?

violet-flowers, 2022-12-03:
Hey, sorry I have not gotten a chance to access the robot or the logs recently. Our build is gradle/android studio based but I believe On Bot Java might still be left in the dependencies file/list (i'm unsure if our TeamCode requires it or not).

Windwoes, 2022-12-03:
But to clarify, you're not actually using the special EOCV bundle file for OnBot, correct?

violet-flowers, 2022-12-03:
I believe not, what should I look for in the code to indicate whether or not we are?

Windwoes, 2022-12-03:
Well, it's a question of whether you followed the [OBJ directions](https://github.com/OpenFTC/EasyOpenCV#installation-instructions-onbotjava) or not.

violet-flowers, 2022-12-03:
We are not using the bundle file for OnBotJava, no. I've gone through and ensured that all the steps for the Android Studio setup where followed properly.

Windwoes, 2022-12-03:
Okay, then yeah I would very much like to see the logs leading up to that failure, because if dynamic loading of the library failed, you should have gotten a popup saying why, before you get to the point of code crashing because it's not loaded.

violet-flowers, 2022-12-03:
I should have access to the robot Monday/Tuesday and will be able to run some tests and send over the logs. What I can say, is that there was no pop up message or any descriptive error message displayed on the driver station. When looking through the logs myself I continuously saw an exception stating that the binary library file (.so) couldn't be found, even though the binary was placed on the internal storage in the top level of the FIRST folder on the robot.

Windwoes, 2023-01-13:
The latest version of EOCV no longer does dynamic loading of the native library from external storage (it's bundled with the app). I'm not sure what was going on here, but it shouldn't happen anymore.

closed 2023-01-13 by Windwoes