OpenFTC/EasyOpenCV #56: FTC Robot Controller App Crashes

TWinston-66, 2022-11-18:
The Robot Controller app stops when running the code below. On the monitor a pop up says `FTC Robot Controller has stopped`. There are no errors on the Driver Station and no errors in the debugger within Android Studio. I copied the `webcamExample` OpMode, and when run it exhibits the same behavior; the FTC Robot Controller crashes on that line. This could be an implementation error on my end...

`webcam = OpenCvCameraFactory.getInstance().createWebcam( hardwareMap.get( WebcamName.class, "webcam1"), cameraMonitorViewId);`

Thanks for any and all help!

Windwoes, 2022-11-18:
Can you provide the robot controller log please?

Windwoes, 2022-11-18:
Actually, I'm going to venture a guess you're using SDK 8.0 and have not updated to the latest version of EOCV which fixes compatibility with SDK v8.0

TWinston-66, 2022-11-18:
That guess would be correct. I should have checked that before asking but thank you! I appreciate the help and thanks for your work!

Windwoes, 2022-11-18:
Great, glad you got it working!

closed 2022-11-18 by Windwoes