OpenFTC/EasyOpenCV #58: When I switch the pipeline, the robot sometimes stops.

allenftc, 2022-12-01:
In my code, I have it move, and then switch pipelines in between move steps. Sometimes, it is fine, but sometimes it causes a noticeable delay (>5 seconds). Do I need to put it on another thread? Thanks for the help.

Windwoes, 2022-12-01:
So, looking at the code, setPipeline is synchronized with the function handling frame processing, meaning that it will block until the current frame has been processed. Does your pipeline take a very long time to process frames? A quick fix would be to spawn a thread to call setPipeline, yes... however I don't like that from an API perspective so I may consider whether I can safely eliminate the synchronization.

Windwoes, 2022-12-01:
I assume that your pipeline was in fact taking a long time to process frames?
In any case I'm going to re-open this for now because I do think the API could be improved.

allenftc, 2022-12-20:
The pipeline was not too slow, it was some contour detection that ran at around 20 fps. I fixed it by having the same pipeline, but changing it to detect different colors.

Windwoes, 2022-12-20:
Do you have an example of the code you were using to switch the pipelines that was causing the issues?

Windwoes, 2023-01-09:
> In any case I'm going to re-open this for now because I do think the API could be improved.

Note: this is fixed on the develop branch https://github.com/OpenFTC/EasyOpenCV/commit/e9067b30b9f83c96177463efe2b730a9561993b7

lukasmwerner, 2023-02-02:
How do we utilize this new bugfix?

Windwoes, 2023-02-02:
I haven't actually done a release that incorporates that commit yet, but I'm hoping to soon.

Windwoes, 2023-02-07:
Fixed in 1.6.0

closed 2023-02-07 by Windwoes