OpenFTC/EasyOpenCV #64: Splitting viewport seems to not work...

alan412, 2023-01-30:
Our code is here: https://github.com/ftc16072/PowerPlay22-23/blob/master/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/ftc16072/OpModes/VisionAutoBase.java

It seems to be the same as the example, but only one of the cameras is seen when we do "show camera stream" on the Driver Station.

Any idea what we are doing wrong?

Thanks in advance,

Alan

Windwoes, 2023-01-30:
Splitting the viewport is only for the preview in the RC activity.
It's only possible to show one stream on the DS at a time; this is an SDK limitation. Feel free to open an issue on the SDK repo.

closed 2023-01-30 by Windwoes

Windwoes, 2023-01-30:
You can however use scrcpy or an external HDMI monitor to see the local view.

alan412, 2023-01-30:
Funny - what I sent to the students: I checked with OpenCV about why we couldn’t see both cameras, and they said we should be able to see them on the RC (We could use scrcpy to see it) but can’t on the driver station due to a limitation in the driver station