OpenFTC/EasyOpenCV #66: App crashes on loading library after upgrade to 1.6.2 and apriltag 1.1.1

IronReignRobotics, 2023-03-07:
We are trying to upgrade to EOCV 1.6.2 with the 8.1.1 ftcsdk. The app crashes as soon as the vision pipeline tries to load. Here are the relevant log lines:

--------- beginning of crash
03-06 19:07:43.816  2538  2692 E AndroidRuntime: FATAL EXCEPTION: OpModeThread
03-06 19:07:43.816  2538  2692 E AndroidRuntime: Process: com.qualcomm.ftcrobotcontroller, PID: 2538
03-06 19:07:43.816  2538  2692 E AndroidRuntime: java.lang.UnsatisfiedLinkError: dlopen failed: cannot locate symbol "_ZN2cv3MatD1Ev" referenced by "/data/app/com.qualcomm.ftcrobotcontroller-1/lib/arm/libEasyOpenCV.so"...
03-06 19:07:43.816  2538  2692 E AndroidRuntime: 	at java.lang.Runtime.loadLibrary0(Runtime.java:989)
03-06 19:07:43.816  2538  2692 E AndroidRuntime: 	at java.lang.System.loadLibrary(System.java:1562)
03-06 19:07:43.816  2538  2692 E AndroidRuntime: 	at org.openftc.easyopencv.OpenCvCameraBase.<clinit>(OpenCvCameraBase.java:965)
03-06 19:07:43.816  2538  2692 E AndroidRuntime: 	at org.openftc.easyopencv.OpenCvCameraFactoryImpl.createWebcam(OpenCvCameraFactoryImpl.java:98)
03-06 19:07:43.816  2538  2692 E AndroidRuntime: 	at org.firstinspires.ftc.teamcode.robots.taubot.vision.provider.AprilTagProvider.initializeVision(AprilTagProvider.java:57)
03-06 19:07:43.816  2538  2692 E AndroidRuntime: 	at org.firstinspires.ftc.teamcode.robots.taubot.PowerPlay_6832.init(PowerPlay_6832.java:227)

any ideas?

Thanks.

Windwoes, 2023-03-07:
Well that is..... puzzling :pig_nose: 

The key bit here seems to be

```
dlopen failed: cannot locate symbol "_ZN2cv3MatD1Ev" referenced by
"/data/app/com.qualcomm.ftcrobotcontroller-1/lib/arm/libEasyOpenCV.so"...
```

The symbol looks to be an OpenCV one.... are you using the default transitive dependency on `'org.openftc:opencv-repackaged-bundled-dylibs:4.5.3-C'`, or are you perhaps using a custom built version of the OpenCV native libraries?

Windwoes, 2023-03-07:
Also what hardware are you running on?

IronReignRobotics, 2023-03-07:
We used to run a custom build long ago, but we've been running EasyOpenCV for the last two seasons. When we revert to 1.5.2 the pipeline works.

Here's what the module dependencies look like:
[image]
 
This is running on a chub with current OS and firmware, and with a Logitech C920 webcam

IronReignRobotics, 2023-03-07:
TeamCode build.gradle looks like:

apply from: '../build.common.gradle'
apply from: '../build.dependencies.gradle'

android {
    namespace = 'org.firstinspires.ftc.teamcode'

    packagingOptions {
        jniLibs.useLegacyPackaging true
    }
}

dependencies {
    implementation project(':FtcRobotController')
    implementation 'org.openftc:easyopencv:1.6.2'
    implementation 'org.openftc:apriltag:1.1.1'
    annotationProcessor files('lib/OpModeAnnotationProcessor.jar')
    implementation 'org.apache.commons:commons-math3:3.6.1'
    implementation 'com.acmerobotics.roadrunner:core:0.5.6'
}

Windwoes, 2023-03-07:
Can you test 1.5.3?

IronReignRobotics, 2023-03-07:
Exact same problem with 1.5.3 as with 1.6.2

IronReignRobotics, 2023-03-07:
Happy to setup a screenshare of the dev system if that helps.

Windwoes, 2023-03-07:
That's very strange because I know multiple people using the latest versions without issue. I might build you an APK that I can prove works on my end and see what happens for you.

Windwoes, 2023-03-08:
Please run `adb shell pm uninstall com.qualcomm.ftcrobotcontroller`

Then try this APK: https://drive.google.com/file/d/14qeWI9PhAZG9I9SZmQderh9UJf-5ARtM/view?usp=share_link

I tested that this APK runs correctly only a Control Hub with a C920 camera.

Windwoes, 2023-03-11:
Oh and if that APK still crashes, then I'd probably suggest trying to factory reset the control hub.

Windwoes, 2023-03-18:
@IronReignRobotics Were you able to test?

Windwoes, 2023-04-18:
Closing due to no response. Feel free to re-open if you still have this issue.

closed 2023-04-18 by Windwoes