OpenFTC/EasyOpenCV #69: Error in OpenCvCameraBase crashes usages of VisionPortal in sdk 9.0

CMP-18996, 2023-09-20:
My control hub logs gave the following:

FATAL EXCEPTION: Thread-10
09-19 19:46:52.102  5726  5903 E AndroidRuntime: Process: com.qualcomm.ftcrobotcontroller, PID: 5726
09-19 19:46:52.102  5726  5903 E AndroidRuntime: java.lang.NoSuchMethodError: No virtual method initActiveOpMode(Ljava/lang/String;)V in class Lcom/qualcomm/robotcore/eventloop/opmode/OpModeManagerImpl; or its super classes (declaration of 'com.qualcomm.robotcore.eventloop.opmode.OpModeManagerImpl' appears in /data/app/com.qualcomm.ftcrobotcontroller-1/base.apk:classes12.dex)
09-19 19:46:52.102  5726  5903 E AndroidRuntime:     at org.openftc.easyopencv.OpenCvCameraBase.emulateEStop(OpenCvCameraBase.java:757)
........
09-19 19:46:52.107  5726  5851 D RobotCore: system telemetry: key=$System$Error$ msg="User code threw an uncaught exception: NullPointerException - Attempt to invoke interface method 'void org.openftc.easyopencv.OpenCvCamera.startStreaming(int, int, org.openftc.easyopencv.OpenCvCameraRotation)' on a null object reference"

Tracking the first error into OpenCvCameraBase.java shows that android studio doesn't understand "initactiveopmode" (last line)

```protected void emulateEStop(Exception e)
    {
        RobotLog.ee("OpenCvCamera", e, "User code threw an uncaught exception");

        String errorMsg = e.getClass().getSimpleName() + (e.getMessage() != null ? " - " + e.getMessage() : "");
        RobotLog.setGlobalErrorMsg("User code threw an uncaught exception: " + errorMsg);

        OpModeManagerImpl mgr = OpModeManagerImpl.getOpModeManagerOfActivity(AppUtil.getInstance().getActivity());
        mgr.initActiveOpMode(OpModeManagerImpl.DEFAULT_OP_MODE_NAME);
..........
```

Ran on a Control Hub from Rev Driver Hub with modified code from the ConceptAprilTag.java.

Windwoes, 2023-09-20:
This looks like an issue due to ABI breakage from the new SDK.
(i.e. EOCV 1.7.0 was compiled against an earlier version of the SDK).

CMP-18996, 2023-09-20:
How would this be fixed? I cloned sdk 9.0 then added the 1.7.0 eocv gradle dependency, is there an error in that?

Windwoes, 2023-09-20:
It's not really something you can fix, I'll have build EOCV against 9.0 and fix it.

That being said, something is crashing that should not be crashing. I.e. even if this issue is fixed, you'd still be getting a crash, it would just be a "nicer" crash.

Do you experience this with the unmodified example code?

CMP-18996, 2023-09-25:
I got around to testing it and found out it was less an eocv issue and more a visionportal issue.

Windwoes, 2023-09-26:
It may well be a vision portal bug that is triggering this, but based on the stack trace, there is absolutely an issue with EOCV that needs to be fixed in the next release.

Windwoes, 2023-12-19:
This has been fixed in 2ac949f which will be included in the next release.

Windwoes, 2024-02-16:
Fixed in FTC SDK v9.1

closed 2024-02-16 by Windwoes