OpenFTC/EasyOpenCV #70: Cannot upload aar file to onbot java of FtcRobotController 9.0

junfengzh, 2023-09-22:
It complains that the aar contains "classes that already exist in the robot controller".

EasyOpenCV version: 1.6.2
FtcRobotController version: 9.0
Hardware: Rev Control Hub

Windwoes, 2023-09-22:
EOCV is now bundled with the SDK as of 8..2

junfengzh, 2023-09-22:
Thanks for quick reply. Did we change the package path? I tried to build a Java file written for last season, and could not resolve classes like org.openftc.easyopencv.OpenCvWebcam.

Windwoes, 2023-09-23:
Package path did not change.

junfengzh, 2023-09-23:
Actually OpenCvWebcam is fine. It is really only org.openftc.easyopencv.OpenCvCameraFactory that is missing in SDK. When I commented out it, the rest of the code built successfully.

I tried to reinstall robot controller apk a couple of times on control hub and still get the same error.

Windwoes, 2023-09-24:
I was able to reproduce this issue on a Nexus 5 with a clean install of the official 9.0 APK. I will look into it.

ahoglund, 2023-09-29:
> EOCV is now bundled with the SDK as of 8.2

This fact should be made clearer on the README, so visitors won't try to add this to their FtcRobotController code

Windwoes, 2023-11-08:
I updated the readme.

closed 2023-11-08 by Windwoes