OpenFTC/EasyOpenCV #71: Camera slow to load

EddieDL, 2023-10-31:
We are using a generic USB camera and we are having issues where it takes a long time for the camera to start the stream. Often we have to stop and start the opmode to get it working or it takes close to a minute to actually start. With a "standard" logitech camera it is quick, but we need the wide field of view of our generic camera.

What can we do to fix this?

Windwoes, 2023-10-31:
Based on my experience with another camera that behaved similarly (Arducam), it's a case of the device responding poorly to a USB protocol reset before trying to start the stream where the fix is to not reset it. Unfortunately that fix must be made in a C++ file inside the SDK itself (which contains the UVC driver) not in EOCV.

Without having a camera on hand it's impossible to say really though.

EddieDL, 2023-10-31:
@Windwoes thanks for the response. Do you know of a camera with a wide for that does work fast or a brand that seems to be better?

EddieDL, 2023-10-31:
https://www.amazon.com/dp/B07R44YHW7?ref=ppx_pop_mob_ap_share this is what we have.

Windwoes, 2023-10-31:
> @Windwoes thanks for the response. Do you know of a camera with a wide for that does work fast or a brand that seems to be better?

Not off the top of my head. Perhaps ask around on the ftc discord or subreddit, or possibly on the ftc community forum.

EddieDL, 2023-10-31:
One thing I forgot to mention is that it is only slow when the robot is just powered on. After it connects one time it works fine. Not sure if that gives anything more. But thought I would mention it.

@Windwoes

Windwoes, 2023-10-31:
Oh, that's interesting. Logs showing an instance of the initial failure and then subsequent working fine might be interesting.

EddieDL, 2023-10-31:
Okay I will see what I can dig up in the logs. I think we can just download them from the Robot Controller. If there is something special that needs to be done, let me know.

EddieDL, 2023-11-02:
We got a new camara and it is WAY faster. Thanks for the responses.

closed 2023-11-02 by EddieDL