OpenFTC/EasyOpenCV #72: CameraFactory does not work after updates

xboxman234, 2023-11-05:
I am sorry if this is the wrong place or way to ask this but before updating my driver controller the webcamExample.java worked find and I was able to get pixels from a matrix but after the updates "import org.openftc.easyopencv.OpenCvCameraFactory" causes an error of cannot find symbol OpenCvCameraFactory or somthing along those lines and I am posting this here because it might be an update issue but if it is not and I am just dumb please point me in the right direciton thank you

Windwoes, 2023-11-05:
This is a known bug in SDK v9.0 specific to OBJ, related to the fact that the EOCV artifact was compiled against a prior version of the SDK which still contained Vuforia. OBJ cries because it doesn't know how to handle that situation and thus excludes the CameraFactory class from the classpath.

Feel free to open an issue on the SDK repo for increased visibility to the community.

Note that this issue is specific to OBJ, it does not affect AS users.

xboxman234, 2023-11-05:
Is there anything I can do? My team is relying on this code for detection because I havent been able to find any other way to get specific pixel values rgb.

Windwoes, 2023-11-05:
You can either switch to using Android Studio, or port your EOCV pipeline to a Vision Processor to be used with the Vision Portal. The LearnJavaForFTC book has a chapter covering that.

Windwoes, 2023-11-06:
(Also it will be fixed in the next SDK release but I don't have a timeline on that)

xboxman234, 2023-11-09:
Sorry for the late question, if I switch to Android Studio can I just copy and paste my code or do I need to import more things?

Windwoes, 2023-11-09:
You should be able to copy paste

Windwoes, 2023-12-19:
This has been fixed in 2ac949f which will be included in the next release

Windwoes, 2024-02-16:
Fixed in FTC SDK v9.1

closed 2024-02-16 by Windwoes