OpenFTC/EasyOpenCV #74: Trying to run OpenCV pipeline at the same time as tensorflow on the same camera results in an error

electricboy6, 2024-08-16:
This is not an issue, more of a request for assistance, not the FTC forums are down. 

Our team is trying to run an OpenCV pipeline on the same frame as a tensorflow one, which obviously results in an error due to the sdk trying to open the camera twice. How would we go about doing this?

Windwoes, 2024-08-16:
Please note that tensor flow will be removed from the SDK in v10.0.

That being said, you have a couple options - you can either rewrite your pipeline to be a vision processor that you can attach to the vision portal, or you can call the TF vision processor from your OpenCV pipeline.

electricboy6, 2024-08-16:
Do you happen to know if the tensorflow calling code is open source? If it is, it would still be legal to use it, correct?

Windwoes, 2024-08-17:
The entire SDK is open source except the driver station. It would be legal to use as long as you didn't have to modify any binary blobs shipped in 10.0.

closed 2024-08-17 by electricboy6